Xastir/Xastir #196: NWS weather Data County Warning

krutzy62, 2022-08-28:
When I run get-NWSdata I get this error:
w_22mr22.dbf
No matching dbfawk signature found.
I think it is trying to match nwsw-ddmmyy-(vv).dbfawk.  I assume that vv is the version so in this case nwsw-ddmmyy-20.dbfawk
Is there an updated .dbfawk for it.  I have a hard time noodling through it.

73, Kevin, W4RKU

tvrusso, 2022-08-28:
Nobody has created a new dbfawk for the new file, and they clearly have changed the signature again.

The current get-NWSdata doesn't download that file version, so you seem to be using a modified version.  When we actually modify get-NWSdata to download new versions, part of the process is to update the dbfawk file.

Where did you get the version of get-NWSdata that is trying to download a 2022 file?  The last time we updated get-NWSdata was in 2020.  It is very out of date.

tvrusso, 2022-08-28:
If you would like to volunteer to update get-NWSdata and provide a valid dbfawk to go with all the current versions of the files, feel free to open a pull request.

To create a new dbfawk file for an updated shapefile whose signature has changed, one simply copies an existing version to a new file name (e.g. copy nws-ddmmyy-20.dbfawk to nws-ddmmyy-22.dbfawk) and modify the dbfinfo line to match the signature present in the file,   One can find out what the signature *should* be with `testdbfawk -D config -d Counties/w_22mr22.dbf 2>&1 | head -3`, which will print out "sig:" followed by a complaint that no matching signature was found.

Then look at the signature and make sure that there is a field called "WFO", the only one we care about.  If it's there, then the only thing one needs to modify in the new dbfawk file is the line that sets "dbfinfo" (it should match the signature of the new shapefile printed above).    At that point you're done and the testdbfawk command above should now say that the signatures match --- and if you replace "head -3" with "less" you'll see line after line of data showing "name" and "key" set to something meaningful.

If there IS no WFO in the signature, then one really needs to work out what the NWS did to change the attributes of the DBF file, and what new attribute corresponds to the one WFO used to refer to (it's the weather office name).  There's a line at the bottom of the dbfawk file that looks for whatever WFO is set to and sets "name" and "key" accordingly --- just change WFO in that line to whatever the new field name is.

krutzy62, 2022-08-28:
I got them from this link: https://www.weather.gov/gis/AWIPSShapefiles.  I updated all them though some were good.
When I first ran get-NWSdata I got a lot of not founds.  Once I updated all the "FIILE" parameters.  Everything worked except the County Warnings.

Yes I ran this: testdbfawk -D config -d Counties/w_22mar22.dbf 2>&1 | head -3
But the result is always: "DBFopen" every time I run testdbfawk I get "DBFopen".  Xastir isn't running.  

I might look at the metadata on the website provided and compare. 

If I get it I will update or provide.  But everything is changing in September 15 so I will wait until then.

Thank you!

krutzy62, 2022-08-29:
Well I tried and failed.  I must be missing something.  I copied nwsw-ddmmyy_20.dbfawk to nwsw-ddmmyy_22.dbfawk.
the 20 version looked like this
dbfinfo="OBJECTID:CWA:WFO:LON:LAT:Region:FullStaId:CityState:City:State:ST";

Here is what the NWS Says:
Shapefile Attributes:
Field Name	Type	Width	Description
WFO	character	3	WFO Identifier (name of CWA)
CWA	character	3	CWA Identifier (same as WFO)
Lon	numeric	10,5	Longitude of centroid [decimal degrees]
Lat	numeric	9,5	Latitude of centroid [decimal degrees]
Region	character	2	NWS Region (Eastern (ER), Western (WR), Central (CR), Southern (SR), Alaska (AR), Pacific (PR)
FullStaId	character	4	Full 4 character internal NWS Station ID for CWA
CityState	character	50	City and state in which the WFO office is located
City	character	50	City designator for office
State	character	50	Full state name in which office is located
St	character	2	Two character state abbreviation for state where office is located
 
So to me the only difference is that the WFO and CWA were swapped,  So I made the 22 version read like this:

dbfinfo="OBJECTID:WFO:CWA:LON:LAT:Region:FullStaId:CityState:City:State:ST";

I don't really know what this means or if it is important:
dbffields="WFO";
and again I don't know why testdbfawk isn't helping
73,
Kevin

tvrusso, 2022-08-29:
I apologize for the bum steer.  I quoted the command that get-NWSdata uses, but it has already changed directory to /usr/local/share/xastir.  That makes the command line I gave wrong for a generic invocation when NOT changed to that directory.

The correct command is `testdbfawk -D /usr/local/share/xastir/config -d /usr/local/share/xastir/Counties/w_22mar22.dbf 2>&1 | head -3`   This assumes you have made the new file in /usr/local/share/xastir/config and made it world readable, and that the dbf file you installed after modifying get-NWSdata is in /usr/local/share/xastir/Counties and is world readable.

The DBOpen error is showing up because it is trying to report that it can't find the file you named after "-d".  It can't find it because I gave you the wrong command.  If you did "head -10" or "less" instead of "head -3" you'd have seen more information about the error.
The thing you're doing wrong is leaving OBJECTID in the dbfinfo.   It also looks like you've got a typo on the "St" field name, which the metadata says is in mixed case but you left as all upper case.  The metadata you posted indicates that it should read:
```
dbfinfo="WFO:CWA:LON:LAT:Region:FullStaId:CityState:City:State:St";
```
The fixed testdbfawk command I just provided should show the error that no matching signature is found before you make that change, and after should say it's all good.

tvrusso, 2022-08-29:
Had some typos in the previous comment that I have fixed --- if you're only reading this based on emails you receive from github, you won't see the fixes.  Look on the github issue page to see the corrected comment.

tvrusso, 2022-08-29:
I also apologize for a second bum steer:  I said that if you had not used "head -3" on the original, erroneous testdbfawk command you would have gotten more information.  I just checked the source code, and that is a lie:  it simply says "DBFopen" with no further information.

I have just changed that.  It now says "DBFopen: could not open dbf file", which I think is a little less cryptic.

krutzy62, 2022-08-29:
Thanks - first I ran the update script to get your new testdbfawk - the message changed.  
I think I had a type on my end too when running by mistyping the .dbf file.
However the change didn't work.  But I did see the DBinfo line which returned:

11 Columns,  125 Records in file
sig: OBJECTID:CWA:WFO:LON:LAT:REGION:FULLSTAID:CITYSTATE:CITY:STATE:ST
No matching dbfawk signature found

So taking that I changed the nwsw_ddmmyy_22.dbfawk to read:
dbfinfo="OBJECTID:CWA:WFO:LON:LAT:REGION:FULLSTAID:CITYSTATE:CITY:STATE:ST";

And that worked.  Unfortunately when I did the update I had to redo the get-NWSdata script but not the ".in version - I don't know enough.  But the change is this:

ease only have ONE of each variable listed here!  Note that the "Valid
# Date" listed on the NWS web pages is the date at which the Shapefile
# should START to be used.  Don't just blindly put the newest filename
# here if that start-date hasn't arrived yet!
#
FILE1="w_22mr22"    # GREEN: NWSM Library: County Warning Area Boundaries
FILE2="z_22mr22"    # GREEN: NWSM Library: Public Forecast Zone Boundaries
FILE3="mz22mr22"    # GREEN: NWSM Library: Coastal and Offshore Marine Zones
FILE4="oz22mr22"   # GREEN: NWSM Library: Coastal and Offshore Marine Zones
FILE5="hz30jn17"    # GREEN: NWSM Library: Coastal and Offshore Marine Zones
FILE6="fz22mr22"    # GREEN: NWSM Library: Fire Weather Zone Boundaries
FILE7="c_22mr22"    # RED: Counties, States, Provinces: U.S. Counties

I have attached both if you want to use them (added txt to the end so they would attach correctly).
[get-NWSdata.txt](https://github.com/Xastir/Xastir/files/9441497/get-NWSdata.txt)
[nwsw_ddmmyy_22.dbfawk.txt](https://github.com/Xastir/Xastir/files/9441498/nwsw_ddmmyy_22.dbfawk.txt)

Thanks for all  of your help.  I finally get it!!

tvrusso, 2022-08-29:
Ok, thanks for taking the time to work through it. 

get-NWSdata in the installed directory is created from get-NWSdata.in in the source tree, so the right one to modify on our end is the .in.  But you have provided enough information for me to get this fixed, and I will try to take care of it this week. 

Again, thank you for working through it and getting something that works.

tvrusso, 2022-08-29:
It also shows that NWS has misstated the contents of their DBF files.  Looks like the only thing they actually changed was that the older ones had mixed-case fields for Region, FullStaId, CityState, City, and State, and the new ones have all upper-case.  That is the only difference between your new dbfawk and the old.

krutzy62, 2022-08-29:
You welcome, it was a learning exercise.  I created a reminder to redo it in September, I will let you know how that goes.  

Yes I noticed that too.  Who knows why.  Probably some new Government standard.  

73, Kevin, W4RKU

tvrusso, 2022-08-29:
Fixed by commit d9d85f7.

Thank you for your help.

closed 2022-08-29 by tvrusso

krutzy62, 2022-09-17:
Tom,

 

Attached is the get-NWS for the ones I saw were to be active 13 SEP 22.  I didn’t see any future ones that I saw.

 

 

73, Kevin, W4RKU

tvrusso, 2022-09-17:
No attachment to be found.

Did you find that any of the new files failed to work with existing DBFAWK files?  or was it as simple as updating file names?

krutzy62, 2022-09-18:
Sorry Tom.  

 

Attached now.

 

Kevin

tvrusso, 2022-09-18:
No, you can't attach to an email to github.  It shows up in the old issue report without the attachment.  It doesn't show up at my personal address witht he attachment either.

Either go open a new issue report on github or describe your changes in text.  The attachment is not so important as a description of what you changed --- was it just the file names downloaded, or did you also have to create a new dbfawk?

krutzy62, 2022-10-11:
No dbfawk didn’t change.  Just the zip file names.  

 

I am sorry I don’t really know how the github stuff works.  I will write it up.  Apologies for the misunderstanding.

 

I will do as you said.

 

73, Kevin, W4RKU

tvrusso, 2022-10-11:
I already updated the get-NWSdata script for the September file name changes in commit e67a502 quite some time ago.  Should be all good now.

commit d9d85f74 (2022-08-29): Update get-NWSdata and associated dbfawk

NWS has changed its data files again, and has made a change to the
field names in the "w" file.  The only change to the field name is
that they're all upper case now, whereas previous versions were mixed
case.

This commit updates get-NWSdata to make it download the new shapefiles
and adds a new dbfawk file to make it work with the new "w" shapefile.

Closes #196