Xastir/Xastir #198: Altitude Appears in Comments

westerkampj, 2022-09-23:
There appears to be a misplaced / in the interface.c code that cause the altitude to not be interpreted correctly and to appear at the beginning of the Comment in an APRS transmission. On line 7521, 

"A=%06ls/",

should be

"/A=%06ld",

I have made and tested this change and the altitude is interpreted correctly and displayed as an Altitude and no longer appears in the Comment.

Thanks!

John W.
W8LRJ

tvrusso, 2022-09-23:
Fascinating.  Yes, that *does* look like a typo, but  it's a bit puzzling that it matters due to the way that this is string is used:

```
      /* get CSE/SPD, Always needed for output even if 0 */
      xastir_snprintf(output_cs,
                      sizeof(output_cs),
                      "%03d/%03d/",
                      my_last_course,
                      my_last_speed);    // Speed in knots

      /* get altitude */
      if (my_last_altitude_time > 0)
        xastir_snprintf(output_alt,
                        sizeof(output_alt),
                        "A=%06ld/",
                        my_last_altitude);
...
```
and then later:
```
        if((strlen(output_cs) < 8) && (my_last_altitude_time > 0) &&
            (strlen(output_alt) > 0))
        {
          xastir_snprintf(output_brk,
                          sizeof(output_brk),
                          "/");
        }

        day_time = localtime(&sec);

        xastir_snprintf(data_txt_save,
                        sizeof(data_txt_save),
                        "@%02d%02d%02d/%s%s%s%s%s",
                        day_time->tm_mday,
                        day_time->tm_hour,
                        day_time->tm_min,
                        my_pos,
                        output_cs,
                        output_brk,
                        output_alt,
                        my_comment_tx);
```

This is *supposed to cause output_cs to end with a "/".  If the output_cs is not of the correct length, *and* the "output_alt" is fresh and non-zero length, then output_brk is supposed to contain "/".  In either case, this snippet is *supposed* to mean that there should be a slash before "A=" no matter what (either because it's already in output_cs or because we add one).

That is not to say I don't think your fix is correct.  It seems, though, to point to a different logic error that was clearly intended to take care of exactly this case.  Perhaps the conditional that sets output_brk to slash is slightly off?  So before I just go ahead and make this change, I'd like to hear from a few other users (paging doctors @we7u and @godfreja).

I'm curious what your packets look like before and after your fix.  Could you cut and paste some before/afters?

godfreja, 2022-10-15:
I haven't fully run this down (too late in evening for that), but I wonder if the problem is cases where we output PHG (I instead of CSG) followed by altitude. From what I have seen of the code so far, there is no trailing / for output_phg like there is output_cs.

tvrusso, 2022-10-15:
Ah.  Yes, that appears plausible.  There is logic to set output_brk if output_phg is less than 6 characters, but that seems to presume that output_phg has a trailing slash if it *is* 6 characters.  And the sprintf for *that* doesn't put the trailing slash in.

Does look like a bug.

It would be nice to see if that's the case that the original poster is running up against.  So I reiterate my earlier request: please cut and paste some packet data that illustrates the problematic packets.

Whether the correct fix is to put the trailing slash into output_phg as is done in all the other cases, or to set output_brk to "/" unconditionally when outputting PHG is another question.

The fix proposed by the original poster appears not to be the universal fix, since in so many other cases the slash is already being accounted for properly.  But it would help to see for certain that this special case really is the one the OP is encountering.

tvrusso, 2023-08-03:
Since we never got a set of clear broken packets to use to diagnose the real source of this error (which according to the discussion looks like it may be one special case not properly putting a slash in where all the other cases *do*), this didn't get fixed and won't be fixed in 2.2.0, scheduled to be released tomorrow.

I would love to get this fixed for the next release, though, so if @westerkampj could be more specific in how to reproduce this bug it would be very helpful.

To be clear:  Xastir does not universally fail to put slashes in the right places, there may only be some very specific special cases where it's not done properly, so the fix suggested in the original report is not the right one.  So we have to nail down exactly what special case is being tripped by your specific use case.

If you could please spend a little time with unmodified code and find a packet you're transmitting that is wrong, then cut and paste that into this report, then do it again with your modified code, it would be a huge help toward getting this issue addressed properly.

westerkampj, 2023-08-03:
Thank you very much for looking over this issue. Here is the packet when I add the "/" before the A (correct result)

2023-08-03 13:36:42 EDT: [W8LRJ](https://aprs.fi/?c=raw&limit=&call=W8LRJ)>APX219,WIDE2-2,qAR,[W8BI](https://aprs.fi/?c=raw&limit=&call=W8BI):=3941.52N/08404.49W-PHG7230John w8lrj@arrl.net

Unfortunately, I am not able to rebuild the code as I have long forgotten which version had the issue and I don't want to install over code that I have working (it appears to be version 2.1.9 and the latest release was 2.1.8). I should have kept better records and recorded the bad packets, I apologize for that. Anyway, when the issue arises, aprs.fi shows the information with "PHG7230John w8lrj@arrl.net" in the comments, rather than as a separate altitude with just the comment "John w8lrj@arrl.net" as it does for the working packet shown above.

Thanks!

John
W8LRJ

tvrusso, 2023-08-03:
Odd, as that packet doesn't have the altitude in it at all.  However it *does* confirm that perhaps the issue is with the PHG output code.

I see that your last transmission was a couple hours ago according to aprs.fi.  I have just adjusted my own station's APRS-IS feed to make sure to display any packets from you, so I can check things like whether PHG is properly displayed.

Could you please make your station transmit again so I can see if I'm receiving you correctly?

tvrusso, 2023-08-03:
Also, did you have a GPS attached to the station when you first noticed the problem?  The issue may be that without an altitude from the GPS we don't attach altitude at all.  Not sure what's going on here yet.

tvrusso, 2023-08-03:
Never mind those requests. 

I just fired up my laptop, added a GPS, enabled PHG, and lo and behold I see exactly the issue originally reported:

Without GPS attached, with PHG disabled:
```
KM5VY-8>APX219,WIDE2-2:=3458.20N/10618.31WlXASTIR-Linux
```

With GPS attached, PHG disabled:
```
KM5VY-8>APX219,WIDE2-2:=3458.19N/10618.31l/A=007298/XASTIR-Linux
```

With GPS attached, PHG enabled:
```
KM5VY-8>APX219,WIDE2-2:=3458.18N/10618.31WlPHG2100A=007277/XASTIR-Linux
```

So the diagnosis is indeed that the issue lies with the funky way we try to force that slash into the altitude string, and how it's completely broken when both PHG is enabled *AND* altitude is available.

I think my solution above, to put the slash where it belongs (in the altitude string) unconditionally, and get rid of the broken "output_brk" logic is the correct fix.  But since we're exactly one day from release of 2.2.0 I will not make this change until after the release, so it can be tested completely before unleashing on the world.

tvrusso, 2023-08-03:
Not shown above, but equally germane, if I have no gps and have PHG enabled (as I always do on my main desktop, fixed station install), PHG is always properly output without any altitude at all (and with no superfluous trailing slash).

Now, we're strangely outputting trailing slashes after course/speed even if there's no altitude string, and this is not correct according to the spec.  There need be no slash between the course/speed and comment string if there's no altitude there, and we're being weird for inserting one.

tvrusso, 2023-08-03:
Adding more detail to the issue title.

"/A=xxxxx" is what is supposed to be placed *anywhere* in the comment string to make compliant APRS clients recognize the station's altitude.  Xastir is putting "A=xxxxx/" in the comments if PHG is being transmitted, and "/A=xxxxx/" if course and speed is being transmitted.

The code responsible for this mess is in interface.c's "output_my_aprs_data" and I believe the solution is [as described earlier](https://github.com/Xastir/Xastir/issues/198#issuecomment-1664271961).

tvrusso, 2023-08-04:
Looked over the code in db.c, and we are in fact only looking for "/A=xxxxxx" in the comments part of received packets, and are correctly looking for it *anywhere* in the comments.  So that is correct, at least.

I will try to get this addressed tomorrow right after I release 2.2.0 and open a pull request for review after I've tested the various bits that I modify.

westerkampj, 2023-08-04:
Wow, Tom. Great work and thank you for tracking this down with the little information I provided.

Thanks!

John
W8LRJ

commit ced6615b (2023-08-06): Merge pull request #213 from tvrusso/issue-198

Fix altitude-in-comment with PHG

Closes #198
closed 2023-08-06 by tvrusso