Xastir/Xastir #201: autoconf

kc2cbd, 2022-12-19:
I can't find an autoconf above 2.69 on Debian Bullseye.

kc2cbd, 2022-12-19:
I think I see the Testing branch (Bookworm) is being used.

tvrusso, 2022-12-19:
Xastir only requires a version above 2.60.  What exactly is the problem?  Does autoconf 2.69 not work for you?

kc2cbd, 2022-12-19:
Bullseye is throwing a error that it autoconf requires 2.71> but 2.69 is the stable version in Bullseye.

tvrusso, 2022-12-19:
Not sure where that could be coming from, since Xastir's configure.ac has this:

```
AC_PREREQ(2.60)
```
We should be happy with anything Debian provides as long as it isn't ancient.  There is nothing in Xastir that would require autoconf 2.71 or higher.

Are you trying to install Xastir from source code or from a debian package?

At what stage in the process does this error show up?

kc2cbd, 2022-12-19:
Getting this when bootstrapping.

<
:~/src/Xastir$ ./bootstrap.sh 
    5) Removing autom4te.cache directory...
    4) Running aclocal...
configure.ac:29: error: Autoconf version 2.71 or higher is required
configure.ac:29: the top level
autom4te: /usr/bin/m4 failed with exit status: 63
aclocal: error: autom4te failed with exit status: 63
    3) Running autoheader...
aclocal.m4:17: warning: this file was generated for autoconf 2.71.
You have another version of autoconf.  It may work, but is not guaranteed to.
If you have problems, you may need to regenerate the build system entirely.
To do so, use the procedure documented by the package, typically 'autoreconf'.
configure.ac:29: error: Autoconf version 2.71 or higher is required
configure.ac:29: the top level
autom4te: /usr/bin/m4 failed with exit status: 63
autoheader: '/usr/bin/autom4te' failed with exit status: 63
    2) Running autoconf...
aclocal.m4:17: warning: this file was generated for autoconf 2.71.
You have another version of autoconf.  It may work, but is not guaranteed to.
If you have problems, you may need to regenerate the build system entirely.
To do so, use the procedure documented by the package, typically 'autoreconf'.
configure.ac:29: error: Autoconf version 2.71 or higher is required
configure.ac:29: the top level
autom4te: /usr/bin/m4 failed with exit status: 63
    1) Running automake...
aclocal.m4:17: warning: this file was generated for autoconf 2.71.
You have another version of autoconf.  It may work, but is not guaranteed to.
If you have problems, you may need to regenerate the build system entirely.
To do so, use the procedure documented by the package, typically 'autoreconf'.
configure.ac:29: error: Autoconf version 2.71 or higher is required
configure.ac:29: the top level
autom4te: /usr/bin/m4 failed with exit status: 63
automake: error: autoconf failed with exit status: 63
Bootstrap complete.
>

I have the latest pull.

tvrusso, 2022-12-19:
That is the most puzzling thing I've seen.  Xastir does NOT require 2.71.  I just pulled a fresh clone and see absolutely nothing that would cause this.

In fact, line 29 of configure.ac, where it's saying that autoconf 2.71 is required, is in fact saying "2.60 is the minimum version we can use" and says nothing about 2.71.  Does your copy of configure.ac in fact have 'AC_PREREQ(2.60)' on line 29?  That's what's in the version in git.

This may be some issue specific to debian and its autoconf packages.  I cannot help you with it as I have no debian system to investigate on, and it is not in fact an Xastir bug.  Xastir's configure is going out of its way to say we can use anything as long as it's newer than 2.60 (which is ancient).  

I suggest you pop on to the Xastir mailing list and query there, as the Debian and Ubuntu package maintainers for Xastir are subscribed to that (or at least used to be) and may be able to give you useful help.

tvrusso, 2022-12-19:
The one thing I *did* try was to change that AC_PREREQ line to require a version of autoconf that is newer than the version I actually have, and it gave the very error that you are getting.  I'm now *really* curious to see if your configure.ac has a modified line 29.

kc2cbd, 2022-12-19:
Yes it does.

tvrusso, 2022-12-19:
"Yes it does."  what?  Have a modified line 29, or have `AC_PREREQ(2.60)`?

`AC_PREREQ(2.60)` does not need to be fixed.

If it doesn't say that, then you've got a modified version and need to toss the modification.  If you're working off a git clone, you want to tell git to toss any local changes.  if you're working out of a tarball then you just need to fix the line so it says what it's supposed to say.

What does "git status" show in your clone, if you're working out of a git clone?

kc2cbd, 2022-12-19:
Yes it has 2.71 in the AC_PREREQ.
I cloned a new directory and it didn't give me the error.

tvrusso, 2022-12-19:
Ah.  So that means you somehow modified that line after cloning.  Damned if I can guess how that could have happened unknowingly.

You don't need to clone fresh, though.  You can always use "git reset --hard" to toss any uncommitted changes and return to the state of the repository before the local mods.  That's why I was asking what "git status" said --- it would probably have said you had unstaged changes or something.

But glad to know that a fresh clone got you back to working.  Can I close this issue now?

kc2cbd, 2022-12-19:
But I followed the Git Instructions and haven't modified anything in that directory.
Thanks

tvrusso, 2022-12-19:
Very strange, as I went ahead and pulled a fresh clone and there was no such issue.  Very strange.

If you still have the old clone, please do me a solid and type "git status" in that directory.  I am now very puzzled.

The last commit to configure.ac in the git repository was from me back in May 2021, and that was to increase the required version from 2.53 to 2.60.  It has *never* been the case that the git repo has had 2.71 there.  I'm stymied how that could have happened without you having changed anything.

kc2cbd, 2022-12-19:
Sorry I deleted it after I was successful with a new clone.

tvrusso, 2022-12-19:
That's a shame.  It'll have to stay a mystery, and there's nothing left to do here.

I can't for the life of me picture how that line got changed, as it has never been that way in github.

I'll just have to close this, since it's resolved by starting fresh.

closed 2022-12-19 by tvrusso

kc2cbd, 2022-12-19:
Yeah I'm not sure either, because I haven't had a reason to edit any files when I do an install.