Xastir/Xastir #202: Remove obsolete autoconf macros

tvrusso, 2023-02-18:
AC_HEADER_STDC, AC_HEADER_TIME, and AC_TYPE_SIGNAL are all deprecated autoconf macros.  Each tries to find standard headers or type information, which in times past might or might not exist.  Nowadays, every single thing tested by these macros has become part of the C standard, and so there should be no system-to-system variation as there once was.

Every one of these macros causes one or more C preprocessor symbols to be defined, and in every case Xastir makes use of those preprocessor macros in ifdefs to decide whether or not to include header files or what type to use for a signal value.  In all modern compilers and systems, those header files always exist and the signal type is standardized, so these macros should not be needed.

After autoconf 2.69, all three of these macros loudly proclaim that they are obsolete and should be removed, and tell users to use "autoupdate" to fix the "problem".  Unfortunately, because Xastir depends on the symbols they produce, this should NOT be done.

The correct fix, which is not difficult but may be time consuming, is to pick through Xastir and remove all the ifdefs that conditionalize includes or type declarations when they use symbols defined by these autoconf macros, making the usage unconditional.  After that, the calls to the macros can be removed.

This will, of course, break Xastir builds on ancient, standard non-compliant systems.  We used to work very hard to keep Xastir running on really old computers that people still had in their garages, storage units, or junkyards.  Xastir is peppered with ifdeffed code that even lets people use ancient versions of our dependent libraries whose APIs kept changing.  But it may be time to let go of that.  Users of ancient systems can just pull an ancient version of Xastir out of git if they need to.

tvrusso, 2023-02-19:
Just to add a little detail related to my last few paragraphs above.

While the quick-and-dirty approach of "remove ifdefs, always assume standard compliance" would address this entire thing for modern systems, as I noted it would break our compatibility with ancient systems that at least at some point in Xastir development was important.  

A more nuanced approach might be to replace the "obsolete" macros with our own that make the same tests and do the same thing, but without the loud, obnoxious commentary about "this is obsolete."

For example, [the current GNU project page on obsolete autoconf macros](https://www.gnu.org/software/autoconf/manual/autoconf-2.70/html_node/Obsolete-Macros.html) has this:

> 
> **Macro: AC_HEADER_TIME**
> 
> This macro used to check whether it was possible to include time.h and sys/time.h in the same source file, defining TIME_WITH_SYS_TIME if so.
> 
> Nowadays, it is equivalent to ‘AC_CHECK_HEADERS([sys/time.h])’, although it does still define TIME_WITH_SYS_TIME for compatibility’s sake. time.h is universally present, and the systems on which sys/time.h conflicted with time.h are obsolete. 

If we remove its usage, we break compatibility with all old systems on which sys/time.h conflicts with time.h.  However, if we do the check ourselves and set TIME_WITH_SYS_TIME correctly we maintain that compatibility.

So perhaps a discussion of how important backward compatibility is to the user base is in order.

godfreja, 2023-02-19:
My thoughts are that xastir is not a rapidly evolving software package and
those running on old operating systems are not going to miss too much by
being stuck on older versions. I consider easing developer burden a more
important goal than extreme backwards compatibility.

I admit that is a self interested viewpoint, but there it is.

73
Jason

tvrusso, 2023-02-19:
I can't say I disagree with that.

Honestly, if one has decided to invest time and energy in maintaining old systems just for the lulz, one can invest that time and energy in porting modern codes back to it.  Not that Xastir is exactly modern, but the principle remains.

But since my goal in opening this issue was to declare that I wasn't able to spend the time even on the simpler approach of tossing backward compatibility, I figured I'd raise the issue for whomever wishes to pick it up.

commit 907feda4 (2023-07-27): Merge pull request #207 from tvrusso/issue-202

Issue 202 -- remove obsolete autoconf macros

Closes #202
closed 2023-07-27 by tvrusso