Xastir/Xastir #205: MAPS

joni73, 2023-06-01:
hi,

I do have problems whit maps. I guess problem is at: EPSG:3067 , https://epsg.io/3067

Firstly dowloadble maps cooridantes are out of system limits ! http://kartat.kapsi.fi/files/taustakarttasarja_jhs180/taustakartta_10k ,.... at http://kartat.kapsi.fi/

Secondly I have not succeed to generate WMSSERVER server request that works:

WMSSERVER
URL https://tiles.kartat.kapsi.fi/taustakartta?service=wms&version=1.1.1&request=GetMap&SRS=EPSG:3067&FORMAT=image/png

I'am trying to use https://kartat.kapsi.fi/ as source of maps, these maps are from
Finish Land Survey material and free to use when origing is mentioned.

Also like to note that whole europe location informations are available due Inspire Act
https://inspire.ec.europa.eu/,...

joni@oh5bzr.rdw.fi

tvrusso, 2023-07-28:
Sorry for having taken so long to respond.

Yes, Xastir does not know how to do transformations between geographic or projected coordinate systems itself.  There is some limited ability to handle UTM coordinates in NAD27 and NAD83, but only for maps in GeoTIFF format, which doesn't help you.  There is NO capability to transform WMS maps, which actually must be in EPSG:4326 (WGS84).

However, the server you provided *can* do that transformation just fine on the server side, and the following .geo file *does* work:
```
WMSSERVER
#
URL https://tiles.kartat.kapsi.fi/taustakartta?service=wms&version=1.1.1&request=GetMap&SRS=EPSG:4326&FORMAT=image/png
```

The important thing is to specify "SRS=EPSG:4326" in there, not EPSG:3067.

This "works" if you're zoomed in close enough to Finland, but as you note *does* produce 

```
convert_from_xastir_coordinates:X out-of-range (too low):18446744073692789480
convert_from_xastir_coordinates:X out-of-range (too high):164376258
```

if you zoom out too far, and shows nothing.

However, if you zoom in close enough, you can get a usable map, but it will be distorted because of the coordinate transformation needed.  For example, if I naively zoom in to bits of Finland near Helsinki, with a typically American ignorance of what parts are actually interesting, I can see varying levels of detail, down to single buildings and streets:

[image]
[image]
[image]
[image]

I hope this is good enough of an answer for you.

tvrusso, 2023-07-28:
I will note, though, that the actual placement of features correctly on the map is entirely dependent on the accuracy of the transformation/warp performed by the WMS server itself.  I expect that in this day in age it is safe to trust that accuracy, but your mileage may vary.

tvrusso, 2023-07-28:
And perhaps a location that might be more interesting to you:

[image]

Unfortunately, as I type this your last APRS positions sent to APRS-IS were about 10-20 minutes ago, and I only just updated my filter for my APRS-IS feed to include you, so I don't have any position reports from you to display on this view.