Xastir/Xastir #214: Performance issues on bookworm

SpudGunMan, 2024-01-20:
Compiling source on a newly updated pi5 with bookworm/wayland performance is very poor 

Application is hanging its functional but unusable at most times.

godfreja, 2024-01-20:
Can you elaborate on what xastir is doing when it performance is bad? Is it drawing maps? If so, what maps are you using?

SpudGunMan, 2024-01-20:
The application "freezes" using default first run worldhi.map

Same behavior using OSM tile 

Xwayland uses 10-20% resources when xastir is freezing up 

Any child windows appear as shrunk windows and need expanded 

The app is fully functioning but appears to be making calls to x/wayland and causing a performance hit

SpudGunMan, 2024-01-27:
saw some related updates (window manager stuff) and tested today the freezing up is a little less no changes to Xastir just the apt packages for rasp-os 

there is still some performance issue it appears but its less painful for sure saw a error flash on console on Xastir launch for 
```convert_from_xastir_coordinates:X out-of-range (too low):18446744073647867136
convert_from_xastir_coordinates:Y out-of-range (too low):18446744073673138816
convert_from_xastir_coordinates:X out-of-range (too high):191284480
convert_from_xastir_coordinates:Y out-of-range (too high):101212800
```

truglodite, 2024-02-03:
Are you accessing locally, via real vnc, or other remote? In my case I noticed major lag in Xastir accessing bookworm with wayland from real vnc. Switching from wayland to xwindows fixed it. Not sure if it's related but thought I'd mention it just in case.

SpudGunMan, 2024-02-03:
Direct console, agree related to Wayland/x11

kc2cbd, 2024-10-31:
This is still an issue in the recently released Raspbian OS (Oct 22, 2024). It apparently doesn't play well with even the new labwc under Bookworm either. Which I think is still using Wayland.

tvrusso, 2026-01-07:
I checked out Xastir on a Raspberry Pi 3b with a brand new vanilla install of Raspberry Pi OS/Debian 13 ("Trixie") and can confirm the OP's report.  Xastir grinds to a halt, taking forever on startup and being very unresponsive to user interactions.

I found the only way to make it usable was to switch from Wayland to X through `raspi-config` (it's in the Advanced options and requires a reboot after selection).

Once I did that switch, it works very well.

I don't know a path forward on this, since Xastir is very explicitly an X program and very tightly tied to X calls.  It is working on Wayland only because there is an X compatibility layer that's at work.  Other than rewriting Xastir to use something like Qt instead of Xt, there is not really much we can do.