Xastir/Xastir #223: Remove TIGER/Line georeferencing

godfreja, 2024-08-02:
In a comment in issue #35 @tvrusso said:

> As an aside, there are other ancient, broken features that probably should get removed, such as the use of the geocoder that once enabled us to process TIGER/line data to create an address database with georeferencing, so you could look up street addresses and have Xastir center on them. Those TIGER/Line files are long obsolete, and the [README.md](https://readme.md/) file even points users at an FTP site for downloading preprocessed geocoder files that is long gone. Another issue should probably be opened to call for clearing that out.

This issue is to track that.

tvrusso, 2024-08-02:
Just for reference:

Address geocoding support was added to Xastir in commit 5f77d06 and subsequent tweaks.

The files that implement the reading of the geocoding database are src/geo-client.c, src/geo-find.c, src/geo.h, src/io-common.c, src/io-mmap.c, and src/io.h.

No code in Xastir actually *creates* the geocoding database, that has to be done by an external program that the commit messages never reference, but which won a Google programming contest (the first one they ran) back around the time this feature was added to Xastir.   (c.f. https://www.cnet.com/tech/services-and-software/google-takes-top-prize-in-its-own-contest/)  It is *possible* that the code could be found in some historical Google page about winning projects, but even if it were, the original TIGER/Line data (in its old topological form, which is wholly unlike the later shapefile format) that it required to work has been discontinued since 2006.  Those files can still be obtained at the Census but would be woefully outdated and useless.

Some Xastir users had gone through the exercise of processing all of the data into useful geocode files and put them on a server, but that site has been down for many years.

I actually still have that file for my state, and just checked --- it still works in Xastir and correctly locates addresses that existed at the time the files were current.  I rather doubt that any current users of Xastir still have those files or use the feature.

tvrusso, 2024-08-03:
Additional tendrils of this feature work their way into db.c, which uses geo.h and extern data defined therein to mark a found destination on the map.  c.f. commit 4858f0f and 0bac4d0, and other commits to db.c that use `destination_coord_lon`, `destination_coord_lat` and other variables in geo.h.

godfreja, 2025-12-12:
Thinking out loud here, I wonder about the possibility of updating to make use of Open Street Map data. Either by querying an online source like [Nominatim](https://nominatim.org/) or by working with local OSM data..

Local would be my first choice since it avoid internet dependency, but online query would be easier, both for us to code and for users. (If we only supported online but made the server name configurable an industrious user could locally run Nominatim.)

tvrusso, 2025-12-12:
I am absolutely certain that having a functional geocoder that doesn't depend on creating geocoding files from data files in a format that the Census doesn't produce anymore using a bit of software that is hard to find anymore would be a benefit to lots of folks.

That sounds like a great thing for you to work on!  😄