Xastir/Xastir #226: New release

AsciiWolf, 2025-03-08:
Last release was in 2023. It would be great if a new version of Xastir could be released so that it still has a chance to make it into the next Debian (Trixie) stable release.

Thanks!

AsciiWolf, 2025-05-25:
New point release would be great if possible.

tvrusso, 2025-05-25:
https://github.com/Xastir/Xastir/releases/tag/Release-2.2.2

closed 2025-05-25 by tvrusso

AsciiWolf, 2025-05-25:
I thought that there were more desktop-related fixes, but probably only these two:

https://github.com/Xastir/Xastir/commit/330e37c7a243fd3f0913431f30fb5b948a1d5b31
https://github.com/Xastir/Xastir/commit/caf30cd6c1c28633dcd078bfa560560a2c90af35

tvrusso, 2025-05-25:
Ah.  Yeah, those were the only desktop related changes.  All the changes I found between 2.2.0 and 2.2.2 are in the release notes.  They were all quite minor.